wouterverweirder/kinect-azure #1: THANK YOU !

JpEncausse, 2019-11-25:
Hello,
Sorry, didn't find your email on github,
Just wanted to say a big **THANK YOU** for this nodejs bridge
It was a real pain in the ass to make work Microsoft SDK / Gesture SDK in C/C++ 
And with this bridge I also get my NodeJS :-)
Just have to check the config options to see if I can reduce the fps or ignore some frames

wouterverweirder, 2019-11-26:
Glad you can use it!

As for the framerate: you can set it to either 5, 15 or 30 (same values as in SDK):

```javascript
kinect.startCameras({
  color_format: KinectAzure.K4A_IMAGE_FORMAT_COLOR_BGRA32,
  color_resolution: KinectAzure.K4A_COLOR_RESOLUTION_1080P,
  camera_fps: KinectAzure.K4A_FRAMES_PER_SECOND_15
});
```

closed 2019-12-05 by JpEncausse

eikaramba, 2020-03-31:
please consider accepting donations for this project ;) would be a first for me but that just shows how awesome it is. i'm literally partying here because i found this repo. i couldn't get my head around c++ .... wuhu!