wouterverweirder/kinect-azure #2: Camera is not mirror image

Chrisbright10, 2019-12-09:
the image returned from the camera is horizontally flipped (e.g. raising the right arm raises the left arm on the screen). Can we get the image to return in a regular mirrored format?

wouterverweirder, 2019-12-22:
Hi Chris,

Thanks for your question. I just reviewed the default Kinect Viewer and I'm not seeing a mirrored image there. Also, the sensor SDK doesn't seem to provide a mirroring option.

If you need mirroring in your viewer. I would suggest doing this through CSS (transforms) or swapping pixel indices when working with the raw feeds?

Chrisbright10, 2019-12-23:
Hi,

I noticed that the latest version of the kinect viewer is not mirrored either, (maybe it changed? Or maybe I am thinking of the Kinect v2 since I switch between them often). 

I will definitely play with the output to try and mirror it. In the meantime, I would love it if you could include an example of mirrored output in the examples! (I am doing mirrored color output with bodytracking) But keep it with the good work! This is an awesome project!

Chrisbright10, 2019-12-23:
The easiest was a CSS transform as shown in this article, thanks!
https://stackoverflow.com/questions/3129099/how-to-flip-images-horizontally-with-html5

closed 2019-12-23 by Chrisbright10