wouterverweirder/kinect-azure #3: Writing video stream

JpEncausse, 2019-12-11:
Hello,

How would you write / append kinect's frame to a video file ? (or ThreeJS canvas to a video file ?)
Would you use a pipe to an ffmpeg stream ?

lisajamhoury, 2020-01-23:
If you [draw the feed to an html5 canvas](https://developer.mozilla.org/en-US/docs/Web/API/CanvasRenderingContext2D/drawImage) you can use a [MediaRecorder() object](https://developer.mozilla.org/en-US/docs/Web/API/MediaRecorder/MediaRecorder) to record the stream. 

Here's an example that uses this library in Electron to record Kinect streams to webm using html5 canvas and mediarecorder: 

```
function createMediaRecorder(html5CanvasID) {
  var idToRecord = html5CanvasID;
  var newMediaRecorder = new MediaRecorder(
    document.getElementById(idToRecord).captureStream()
  );
  var mediaChunks = [];

  newMediaRecorder.onstop = function(e) {
    // The video as a blob
    var blob = new Blob(mediaChunks, { type: "video/webm" });

    // Reset Chunks
    mediaChunks.length = 0;

    // Display the video on the page
    // var videoElement = document.createElement('video');
    // videoElement.setAttribute("id", Date.now());
    // videoElement.controls = true;
    // document.body.appendChild(videoElement);
    // videoElement.src = window.URL.createObjectURL(blob);

    var fs = require("fs");
    try {
      fs.mkdirSync(recordingLocation);
    } catch (evt) {
      if (evt.code != "EEXIST") throw e;
    }

    // Read the blob as a file
    var reader = new FileReader();
    reader.addEventListener(
      "loadend",
      function(e) {
        // Create the videoBuffer and write to file
        var videoBuffer = new Buffer(reader.result);

        // Write it out
        var filename = recordingLocation + id + recordStartTime + ".webm";
        fs.writeFile(filename, videoBuffer, function(err) {
          if (err) console.log(err);
          alert("Your file has been saved to " + filename);
        });
      },
      false
    );
    reader.readAsArrayBuffer(blob);
  };

  // When video data is available
  newMediaRecorder.ondataavailable = function(e) {
    mediaChunks.push(e.data);
  };

  // Start recording
  newMediaRecorder.start();
  return newMediaRecorder;
}

```

If you're looking for an one-off out of the box solution to record a Kinect stream to a video file, I built an app that uses this library that has record functionality for Windows and Azure Kinect streams. https://kinectron.github.io/docs/server-windows.html#recording

JpEncausse, 2020-01-23:
Hello,
Can't share the code for now,but it works very well
- receive the color frame from kinect
- wrap it into Jimp to fix some content (blur faces)
- forward to ffmpeg and write it toa  video file

wouterverweirder, 2020-01-29:
Thanks for sharing the example @lisajamhoury !

closed 2020-01-29 by wouterverweirder