wouterverweirder/kinect-azure #7: Question: Timeline for User Masking?

lisajamhoury, 2020-01-23:
Thanks again for this amazing library! I've had a request for user masking in Kinectron. I wanted to see if you have a timeline for when you will be including it in this library. No rush, just checking.

wouterverweirder, 2020-01-29:
I'll give the library some attention next week - hope to get to user masking.

wouterverweirder, 2020-02-16:
Did not get to this feature yet. However, @grumpy-sailor added a bunch of new features (flipping BGRA to RGBA on the C side, doing the depth coloring (blue to red) on the C side and: **masking pixels based on the distance from the sensor**. There is an example at https://github.com/wouterverweirder/kinect-azure/blob/master/examples/electron/renderer/demos/color-stream-rgba32_alpha_applied.html

wouterverweirder, 2020-04-10:
Just added body index maps, in depth and color space.

You can see demos here:
- [display the raw body index map feed](https://github.com/wouterverweirder/kinect-azure/blob/master/examples/electron/renderer/demos/body-tracking-body-index-map.html)
- [mask people in the color image](https://github.com/wouterverweirder/kinect-azure/blob/master/examples/electron/renderer/demos/body-tracking-body-index-map-color-mask.html)

closed 2020-04-10 by wouterverweirder

lisajamhoury, 2020-04-11:
Awesome. Excited to try it out. Thanks!