wouterverweirder/kinect-azure #8: Upgrade Body Tracking package to 0.9.5

Chrisbright10, 2020-01-23:
It seems in the settings that the body tracking package is on 0.9.4 and they made some big improvements in the newest version, could you upgrade the package version? 

FYI - I tried to do this locally and found my app crashing on the createTracker() function call, so as a heads-up you might have some issues with that.

fractalfantasy, 2020-01-27:
Body Tracking package 1.0.0 was just released! Would love to see it incorporated, in particular CPU body tracking mode (for those who don't have NVIDIA gpus)

wouterverweirder, 2020-01-29:
Good call - should be an easy one to implement. I've put it on my todo list for next week.

JpEncausse, 2020-01-29:
> Body Tracking package 1.0.0 was just released! Would love to see it incorporated, in particular CPU body tracking mode (for those who don't have NVIDIA gpus)

Yes that would be great to have the choice between CPU and GPU.

Did you try if there is better perfs ?

wouterverweirder, 2020-02-02:
Just published an update (version 0.0.9) which uses the Body Tracking SDK v 1.0.0 - let me know how this works for you :-)

fractalfantasy, 2020-02-08:
Amazing thanks Wouter! Just travelling but will test on my kinect when I get home :)

fractalfantasy, 2020-02-15:
So i've just tested on my system and it seems the Body Tracking example is still calling for the cuda drivers. I looked into the code and didnt see any CPU mode variable. Did you happen to implement this? 

Here's the command line error:

> [2020-02-15 13:03:47.845] [error] [t=9328] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (157): Create(). C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:97 onnxruntime::CudaCall C:\agent\_work\11\s\onnxruntime\core\providers\cuda\cuda_call.cc:91 onnxruntime::CudaCall CUDA failure 35: CUDA driver version is insufficient for CUDA runtime version ; GPU=4194304 ; hostname=DESKTOP-2QHPQD0 ; expr=cudaSetDevice(device_id_);
> 
> 
> [2020-02-15 13:03:47.846] [error] [t=9328] [K4ABT] D:\a\1\s\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
>

wouterverweirder, 2020-02-16:
Sorry about that, I didnt check the docs and though it would fallback to cpu automatically.. Just released a new version (v0.0.11) which adds tracking options, including CPU mode.

You can now pass options when creating the tracker:

```javascript
kinect.createTracker({
  processing_mode: KinectAzure.K4ABT_TRACKER_PROCESSING_MODE_CPU
});
```

Other options are the sensor orientation and the gpu device id:

```javascript
kinect.createTracker({
  sensor_orientation: KinectAzure. K4ABT_SENSOR_ORIENTATION_FLIP180,
  gpu_device_id: 2
});
```

You can check a demo at https://github.com/wouterverweirder/kinect-azure/blob/master/examples/electron/renderer/demos/body-tracking-skeleton-2d-cpu.html

fractalfantasy, 2020-02-16:
amazing thank you, everything is building great now!

The CPU body tracking is extremely slow for me (about 2 FPS on the body tracking side), are you getting the same over there? was hoping it would be at least as fast as the kinect v2 body tracking

fractalfantasy, 2020-02-16:
here's a little video to illustrate better:
https://www.dropbox.com/s/pm7t5l04mxk9w26/CPU-Body-Tracking-Azure-Kinect.mov?dl=0

All cores of my CPU (Intel Core i7) seem to be running at full tilt so I don't think its a multi-threading issue.

wouterverweirder, 2020-02-17:
Yes, I have the same, slow framerate on my system. I'm not on my windows system right now, but a test could be to compare it with the performance of one of microsoft's examples: https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/simple_3d_viewer

fractalfantasy, 2020-02-17:
Thanks, I just tested the CPU body tracking from sdk, and as you suspected it's running slow there aswell.. here's a video:
https://www.dropbox.com/s/j1cnjephldp95vr/Azure-Kinect-BodyTrackingv1.0-CPU-Test.MOV?dl=0

I suppose it's up to the K4A dev team to fix this!

closed 2020-02-18 by wouterverweirder