wouterverweirder/kinect-azure #9: Manual Exposure

JpEncausse, 2020-01-28:
How can I set the Kinect Exposure to Manual ?
I see on the recorder there is an option, but I don't see / understand how to use it in the bindings
Thanks !

wouterverweirder, 2020-01-29:
It's not in this library yet - but will add it to my todo list for next week :-)

JpEncausse, 2020-01-29:
Thanks great !

JpEncausse, 2020-02-08:
> It's not in this library yet - but will add it to my todo list for next week :-)

If you have time would be awesome !

My usecase is to perform computer vision on image stream. But dynamic exposure change the color of things (even with HSL) and it becomes very hard to track an object.

Thanks !

wouterverweirder, 2020-02-16:
Not done yet, for my own reference: will need to check https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gae81269489170b26b7f0bbe1a7f9d31d6.html#gae81269489170b26b7f0bbe1a7f9d31d6

JpEncausse, 2020-02-17:
Yes the implementation on the SDK recorder tool args is :
- a boolean yes/no to activate the autofocus
- an int with the exposure value

wouterverweirder, 2020-04-09:
added in version 0.0.13, thanks to @peteshand's work - also added a demo (color controls) showing this new feature.

closed 2020-04-09 by wouterverweirder