wouterverweirder/kinect-azure #10: npm i error

peteshand, 2020-01-30:
I'm getting the following error when running npm i
Has anyone else see something similar?

////////////

C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  Copying C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master/sdk/sensor-sdk/lib/native/amd64/release/k4a.dll to C:\Users\TAC-Brix1\Downloads\kinect-azure-ma
  ster\kinect-azure-master/build/Release\k4a.dll
          1 file(s) copied.
  Copying C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master/sdk/sensor-sdk/lib/native/amd64/release/depthengine_2_0.dll to C:\Users\TAC-Brix1\Downloads\kin
  ect-azure-master\kinect-azure-master/build/Release\depthengine_2_0.dll
          1 file(s) copied.
  Copying C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master/sdk/body-tracking-sdk/lib/native/amd64/release/k4abt.dll to C:\Users\TAC-Brix1\Downloads\kinect
  -azure-master\kinect-azure-master/build/Release\k4abt.dll
          1 file(s) copied.
  kinect_azure.cc
  win_delay_load_hook.cc
..\src\kinect_azure.cc(25): error C2039: 'ThreadSafeFunction': is not a member of 'Napi' [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master\build\kinectAzu
re.vcxproj]
  c:\users\tac-brix1\downloads\kinect-azure-master\kinect-azure-master\node_modules\node-addon-api\napi-inl.h(15): note: see declaration of 'Napi'
..\src\kinect_azure.cc(25): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect
-azure-master\build\kinectAzure.vcxproj]
..\src\kinect_azure.cc(25): error C2146: syntax error: missing ';' before identifier 'tsfn' [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master\build\kinect
Azure.vcxproj]
..\src\kinect_azure.cc(164): error C2065: 'tsfn': undeclared identifier [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master\build\kinectAzure.vcxproj]
..\src\kinect_azure.cc(164): error C3083: 'ThreadSafeFunction': the symbol to the left of a '::' must be a type [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure
-master\build\kinectAzure.vcxproj]
..\src\kinect_azure.cc(164): error C2039: 'New': is not a member of 'Napi' [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master\build\kinectAzure.vcxproj]
  c:\users\tac-brix1\downloads\kinect-azure-master\kinect-azure-master\node_modules\node-addon-api\napi-inl.h(15): note: see declaration of 'Napi'
..\src\kinect_azure.cc(164): error C3861: 'New': identifier not found [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master\build\kinectAzure.vcxproj]
..\src\kinect_azure.cc(488): error C2065: 'tsfn': undeclared identifier [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master\build\kinectAzure.vcxproj]
..\src\kinect_azure.cc(488): error C2228: left of '.BlockingCall' must have class/struct/union [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master\build\kin
ectAzure.vcxproj]
  ..\src\kinect_azure.cc(488): note: type is 'unknown-type'
..\src\kinect_azure.cc(510): error C2065: 'tsfn': undeclared identifier [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master\build\kinectAzure.vcxproj]
..\src\kinect_azure.cc(510): error C2228: left of '.Release' must have class/struct/union [C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master\build\kinectAz
ure.vcxproj]
  ..\src\kinect_azure.cc(510): note: type is 'unknown-type'
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\TAC-Brix1\Downloads\kinect-azure-master\kinect-azure-master
gyp ERR! node -v v10.15.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE

peteshand, 2020-02-05:
@wouterverweirder what version of Node and Windows are you running?

wouterverweirder, 2020-02-05:
@peteshand I'm on node version 11.13.0. Just tested with your node version (10.15.3) and I get the same compile error. I would suggest upgrading your node version to 11+ if possible. Will need to either look into fixing this for older versions, or bumping the minimum required node version of this library.

peteshand, 2020-02-05:
Thanks for looking into this. I'll see if upgrading resolves the issue tomorrow.

peteshand, 2020-02-06:
I can confirm I've upgraded to 12.14.1 and am no longer experiencing the error.

closed 2020-02-06 by peteshand