wouterverweirder/kinect-azure #17: Sync between bodyFrame and colorFrame

JpEncausse, 2020-02-20:
Hello,

It seems there is a lag between colorFrame and bodyFrame. 
It's like the bodyFrame is few frame late compared to colorFrame.

- I don't think it comes from the JS code because I work on provided "data" with all frames
- Did someone also see that behavior ? (I paint the skeleton on the color frame)
- Is there a kind of sync mechanism like in the old Kinect 2 ?

Thanks !

wouterverweirder, 2020-02-20:
Have you tried with passing the synchronized_images_only option in the startCameras method?

```javascript
kinect.startCameras({
  depth_mode: KinectAzure.K4A_DEPTH_MODE_WFOV_2X2BINNED,
  color_format: KinectAzure.K4A_IMAGE_FORMAT_COLOR_BGRA32,
  color_resolution: KinectAzure.K4A_COLOR_RESOLUTION_720P,
  camera_fps: KinectAzure.K4A_FRAMES_PER_SECOND_30,
  synchronized_images_only: true
});
```

JpEncausse, 2020-02-25:
I give it a try after upgrading to latest version, it seems there is a slight lag when moving the hands fast but it's better.

Thanks ! Didn't know that option

closed 2020-02-25 by JpEncausse