wouterverweirder/kinect-azure #18: Wrong order of joints

eikaramba, 2020-03-31:
I don't know why exactly, but somehow the order of the joints is wrong. See the following comparison

<pre>
joint           index in kinect-azure      index according to documentation
right ear                           0                                   31
pelvis                              1                                    0
spine naval                         2                                    1
.....
left eye                           29                                   28
left ear                           30                                   29
right eye                          31                                   30
</pre>

I observed this because i wanted to simply paint the pelvis green in the Body Tracking 2D example:
<pre>
body.skeleton.joints.forEach((joint,index) => {
        if(index==KinectAzure.K4ABT_JOINT_PELVIS){
                      outputCtx.fillStyle = 'green';
        }else{
                      outputCtx.fillStyle = 'red';
        }
        outputCtx.fillRect(joint.colorX, joint.colorY, 10, 10);
});
</pre>

But this code was not working, so now I'm using a helper method to fix the order
```
        function fixOrder(index){
          if(index==0){
            return 31;
          }else{
            return index-1;
          }
        }
```

I also looked at the kinect_azure.cc file, however at a first glance it looks okay.

wouterverweirder, 2020-04-09:
Good catch - got the same behavior. Will investigate.

wouterverweirder, 2020-04-10:
Looks like it's the 2D transformed points that are linked to the wrong joints, the 3D coords are correct.

wouterverweirder, 2020-04-10:
Looks like it had to do with rendering joints in color space while there isn't a color feed open. My demo was a bit confusing.

- Updated library to version 0.0.15, only exposing the color coords when the color feed is actually open.
- Updated the 2d joints demo to render the depthX and depthY coords.
- Added an additional demo, showing skeleton overlay on color feed.

commit d4074e80 (2020-04-10): fix joint coords in 2d space - closes #18
closed 2020-04-10 by wouterverweirder