wouterverweirder/kinect-azure #20: Compatibility with SDK1.4

JpEncausse, 2020-04-05:
Hello,

Would it be possible to add compatibility 
- with [Kinect SDK 1.4](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CHANGELOG.md) ?
- and [Kinect Body SDK 1.0.1](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download)

Many thanks !

wouterverweirder, 2020-04-09:
updated the SDK to version 1.4 in release 0.0.14 (hey, my patch version number kind of matches the sdk version)

the body SDK is still on 1.0.0 - for some reason 1.0.1 isn't working here - onxruntime does not want to launch...

wouterverweirder, 2020-04-10:
Body tracking SDK is now on 1.0.1, it's in release 0.0.16. If running in an existing project, you might want to manually delete:
- the dll files in your project root
- the onnx file

commit a344673d (2020-04-10): updated body tracking SDK to 1.0.1 - closes #20
closed 2020-04-10 by wouterverweirder