wouterverweirder/kinect-azure #25: body tracking not working with recent sdk?

ikenichiro, 2020-04-10:
running the following sample runs the Demos.
$ cd examples/electron
$ npm install
$ npm start
However, all the body tracking seems not to be working returning the following error example.
>  [2020-04-11 05:11:35.779] [error] [t=17784] [K4ABT] D:\a\1\s\src\TrackerHost\TrackerHost.cpp (157): Create(). Find onnxruntime.dll at                                      
> [2020-04-11 05:11:35.779] [error] [t=17784] [K4ABT] D:\a\1\s\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create() 

In the folder examples/electron, I can confirm the dll's and dnn_model_2_0.onnx files. 

Although the demo application doesn't crash, there are no tracking info displayed on the streaming images.

wouterverweirder, 2020-04-11:
Hmm, damn - getting the same issue here. 1.0.1 seems to be a bit wonky. Will investigate and rollback to 1.0.0 if needed.

wouterverweirder, 2020-04-11:
I think I'm onto something, looks like they messed up dll lookup paths when running apps...

wouterverweirder, 2020-04-11:
Published a new version with a fix (0.0.19). The body tracking dlls need to be in this module's build directory as well as the application root for some reason. Try this newer version and let me know how it works for you?

ikenichiro, 2020-04-11:
Thank you for the fast fix.
Installing 0.0.19 fixed the problem for me too.

wouterverweirder, 2020-04-11:
Good to hear - thanks for your other pull request as well!

closed 2020-04-11 by wouterverweirder