wouterverweirder/kinect-azure #28: Keep getting handler took too long violations

Chrisbright10, 2020-06-16:
I am currently trying to use this package and seem to be getting things like "[Violation] 'finish' handler took 3111ms". I have a dialog that turns the camera off on opening and back on during its closing callback and this keeps causing my program to crash. Specifically the open camera function seems to take ~3000-4000ms each time. Is there any way we can work around this or see what we can do to speed that up? For my use case I need the code to be synchronous so that I can ensure the camera has started/stopped, but i also need it to be fast enough to not crash my program waiting.

wouterverweirder, 2020-07-04:
Hi Chris,

Thanks for your report. Could you provide me with a demo app, isolating this problem?

Chrisbright10, 2020-07-18:
I have been using it inside of an Angular app integrated with electron. I have found a workaround for this by not using popup modal dialogs with callbacks (so the callbacks don't timeout). But it might take me some time to provide an app shell with an example (my app is rather large and complex and proprietary).

wouterverweirder, 2020-07-22:
Thank you. These are edge cases which would take me a while to reconstruct. Appreciate your help!

stevenrick, 2020-08-10:
@Chrisbright10 might be worth digging into the [Azure Kinect SDK](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK) because the issue at hand (from what I could glean) has less to do with this library and more to do with your usage pattern of turning the camera on and off. We were doing similar things and found that while you can get accurate status on device open or start camera (in part thanks to #19), the internal functions k4a_device_stop_cameras() and k4a_device_close() both are void returns, meaning the best you can do is check that the pointer to the device (device handle) is null. But the actual device stop and release happens deeper at a driver and OS level. The Azure Kinect is marked as free but does not actually get released until the lower system calls resolve. And this could take no time or a more significant amount of time (a couple of seconds) depending on many factors outside of your control. The longest we've seen is 8 seconds. Typically closer to 1 second.

To work around this you can instead open the device and simply not do anything with the `data ` feed coming from the listener
`kinect.startListening((data) => { ... }` when you don't need it. It's not the most efficient approach but if you need a timely approach to grabbing select bits of data it is better to leave it open and read when you need it than to keep opening/closing it and waiting

closed 2021-04-29 by wouterverweirder