wouterverweirder/kinect-azure #40: Body Tracking v1.1.0 update?

fractalfantasy, 2021-03-23:
any plans of updating this node to use the latest Body Tracking SDK?

wouterverweirder, 2021-04-25:
Ah - will look into this. Looks like they haven't published it on nuget for some reason, so I'll need to change my build proces. Might move to precompiled binaries as well.

fractalfantasy, 2021-04-28:
That’s great news!  The new body tracking update looks like quite a bit of an improvement from the first ones. They added support for all GPUs, and better performance overall. 

About installing directly without NuGet, I got this answer on the official forum, might be useful:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1550#issuecomment-825396669 <https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1550#issuecomment-825396669>

wouterverweirder, 2021-04-28:
Thanks for sharing - got some time on a Windows machine tomorrow. Another interesting add with the new SDK is the option to specify the onnx location: https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download#v110

wouterverweirder, 2021-04-29:
Published a new version (0.1.0) which uses the Body Tracking SDK 1.1.0

There are two potential breaking changes:

1. You will need to install the [Azure Kinect Body SDK](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download). The Body Tracking SDK relies on quite a few dll & onnx files, which you can find in the `Azure Kinect Body SDK\tools` directory. Either add that folder (eg C:\Program Files\Azure Kinect Body Tracking SDK\tools) to your PATH or copy the dll and onnx files into the root of your project.
2. The default setting for GPU tracking no longer works on my system (only CUDA works, not the TensorRT or DirectML). This might be something on my system. You'll need to pass in the preferred processing mode when creating a tracker:

```javascript
  kinect.createTracker({
    processing_mode: KinectAzure.K4ABT_TRACKER_PROCESSING_MODE_GPU_CUDA
  });
```

You can also specify the model path, if you don't want to use the default model:

```javascript
  kinect.createTracker({
    processing_mode: KinectAzure.K4ABT_TRACKER_PROCESSING_MODE_GPU_CUDA,
    model_path: "C:\\Program Files\\Azure Kinect Body Tracking SDK\\tools\\dnn_model_2_0_lite_op11.onnx"
  });
```

wouterverweirder, 2021-04-29:
Just pushed another update - v0.1.1 is now using precompiled binaries

fractalfantasy, 2021-04-29:
[image]

Running into this install error, is there something additional I need to download before running npm install?

wouterverweirder, 2021-04-29:
Ah - made a typo creating the release - removed the "v" from the release name - could you try again?

fractalfantasy, 2021-04-29:
yes just pulling the latest now

fractalfantasy, 2021-04-29:
It installed with the precompiled binaries :) 

The direct ML/TensorRT isn't initiating on my system either. 

 On the _Body Tracking 3D_ electron example I switched processing mode to Direct ML using this call from the [documentation](https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.1.x/group__btenums_ga9d8d849736a88894ccc9b1356fd91bee.html): 
KinectAzure.K4ABT_TRACKER_PROCESSING_MODE_GPU_DIRECTML

got this error:

```
Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFCD2E8F2AC: (caller: 00007FFCD2EDEE66) Exception(3) tid(366c) 80070057 The parameter is incorrect.

[2021-04-29 14:36:13.814] [error] [t=13932] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (185): Create(). Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFCD2E8F2AC: (caller: 00007FFCD2EDEE66) Exception(3) tid(366c) 80070057 The parameter is incorrect.

[2021-04-29 14:36:13.815] [error] [t=13932] [K4ABT] ..\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()

```

wouterverweirder, 2021-04-29:
Yes, same here - only the CUDA mode seems to work.

wouterverweirder, 2021-04-30:
It looks like pyKinectAzure is struggling with the same bug / issue: https://github.com/ibaiGorordo/pyKinectAzure/issues/20

ibaiGorordo, 2021-04-30:
Hi, I was able to fix the problem by just loading the **directml.dll** library in python, in my case I ussed ctypes library, but if you are able to load .dll files, it should work.

Good luck with your project!

wouterverweirder, 2021-04-30:
Thanks for the suggestion @ibaiGorordo - will look into this approach!

fractalfantasy, 2021-05-07:
> Looks like kinect-azure node is referring to System32/directml.dll that comes with Windows 10 installation.
> I replaced the system's directml.dll with the one from BT SDK 1.0.1 and voila! DirectML got working.
>
> I wonder is there a way to explicitly tell NodeJS module where to look for DLL's it referring to?

according to Microsoft documentation if the systems directml.dll is already loaded into memory it’ll default to that rather than using local one

https://docs.microsoft.com/en-us/windows/win32/dlls/dynamic-link-library-search-order

mikkamikka, 2021-05-07:
@wouterverweirder @fractalfantasy Ok, I think I figured out easiest solution. Just drop a copy of directml.dll to your `..\kinect-azure\examples\electron\node_modules\electron\dist` directory. 
Next time electron.exe starts it finds directml.dll just next to it.
Hope it helps!

wouterverweirder, 2021-05-07:
Thanks for the suggestions. I think it's best to keep the dll resolving a responsibility of the OS (aka: avoiding overriding this inside of this module's source code). Placing it next to the executable should probably do the trick. Will test when I get my hands on a windows machine next week (my day to day computer is a mac 😅).

wouterverweirder, 2021-07-14:
Closing, I've updated the README, mentioning where to find the DLL files and what to do with them: https://github.com/wouterverweirder/kinect-azure#installation

closed 2021-07-14 by wouterverweirder