wouterverweirder/kinect-azure #46: Typescript support (d.ts)

challenger71498, 2021-06-24:
Above all, thank you for creating this amazing asset.

# Issues

Since I am developing with Typescript environment, I need a declaration file to specify the type of data output when listening.

# Solution

I made my own d.ts temporarily [here](https://gist.github.com/challenger71498/f4756d3f12cc19ceb291ed5ca9fb5e52), so if anyone needs type definition can use this.

I would like to contribute this file, but I don't really know how to. Should I PR this? Or create a @types/kinect-azure sth like that?

wouterverweirder, 2021-07-07:
Hi there,

Thanks for that! Haven't worked with Typescript yet, but I can see how this is useful. Will look into this, if this doesn't add extra dependencies to this project, I'm all up for adding the type definitions to this repo.

challenger71498, 2021-07-28:
Thanks for quick reply!

Nope, it has no dependencies at all, so totally fine for adding it on the repo!

wouterverweirder, 2021-07-29:
Cool - could you create a pull request, with your kinect-azure.d.ts in the lib folder (next to Kinect-azure.js) and I'll merge it.

challenger71498, 2021-07-30:
Thank you for quick reply!

I just created a PR on it :) glad that I can contribute to this amazing repo!

wouterverweirder, 2021-07-30:
Thanks a lot for contributing this! Merged :-)

closed 2021-07-30 by wouterverweirder