wouterverweirder/kinect-azure #47: Package will not build with Electron Builder

Chrisbright10, 2021-06-30:
I have just migrated over to this package from the kinect2 package and am having trouble building the package for production. I keep getting errors that it can't find dependencies and it is looking for the package.json to help resolve it. I wasn't getting this error with the kinect2 package before and I suspect it has to do something with all of the extra needed files for the new kinect-azure package. 

I am also getting lots of issues because of the use of require.js, since I am using a modern frontend framework to load in the camera package I am having to change the require() to window.require, which doesn't seem all that sustainable. Is there a better way to lad in the deps?

wouterverweirder, 2021-07-07:
Hi Chris,

I've put together a basic demo application with electron-builder, electron-webpack and kinect-azure (with Body Tracking): https://github.com/wouterverweirder/electron-builder-kinect-azure-demo

An important addition to the package.json config, is copying the necessary DLL files for body tracking to work in the release build:

```json
  "build": {
    "appId": "be.aboutme.kinect-azure.electron-builder-kinect-azure-demo",
    "win": {
      "extraFiles": [
        {
          "from": "C:\\Program Files\\Azure Kinect Body Tracking SDK\\tools",
          "to": "./"
        }
      ]
    }
  },
```

Chrisbright10, 2021-07-07:
I attempted to pull down the project, it isn't working for me, here is the error when i try and build dev

[image]

wouterverweirder, 2021-07-09:
You'll get this error when windows can't find the necessary dll files, even when the kinectAzure.node file is there.

I have added an extra script to the package.json file of [the demo project](https://github.com/wouterverweirder/electron-builder-kinect-azure-demo).

Get the latest version, and run

```
npm run copy-dlls
```

before launching in dev mode. I'm assuming the dll files are in the default install directory of the body tracking sdk, aka `C:\Program Files\Azure Kinect Body Tracking SDK\tools`

A distribution build should have worked, as the dll copying is included in the electron-builder config.

Chrisbright10, 2021-07-12:
I ran it and wasn't able to get it to run, ended up having the same issue even though the dlls successfully copied over

[image]

[image]

Chrisbright10, 2021-07-12:
I tried downgrading the kinect-azure package to 0.0.19 and that did seem to run, but wasn't able to build a release package.

wouterverweirder, 2021-07-14:
Ok, I've just pushed some changes on how the prebuild binaries of this module are handled. Try using version 0.2.2 of this library? If not, could you post the full log of what happens when you npm install this module?

Chrisbright10, 2021-07-15:
One quick fix... the xcopy command wasn't working for me, I looked into it and fixed it... here is the corrected command so that others wont have the same issue I did with that part. I am currently working on the rest, I will update you when I have tried upgrading.

"xcopy \"C:\\Program Files\\Azure Kinect Body Tracking SDK\\tools\" \".\" /H /C /Y"

Chrisbright10, 2021-07-15:
I still can't get it to build with the new format. below is the yarn install

[image]

next is the copy-dlls

[image]

and the rest of the yarn dev

[image]

Chrisbright10, 2021-07-15:
Do you think it might have anything to do with this part of the error message? I have never seen that before....

[image]

wouterverweirder, 2021-07-19:
I am assuming the kinectAzure.node file is on your system? Could you open that file with dependency walker (https://www.dependencywalker.com) and check what dll files it is reporting missing?