wouterverweirder/kinect-azure #50: Has anyone gotten this to work in the browser?

leskodan, 2021-07-27:
I am not familiar with electron, but I assume that there is access to the OS that makes the integration between node.js, the Kinect, and a GPU doable. I would be very curious to know if anyone has gotten this to work in a browser? Is it possible to get access to the Kinect and SDK without an application running on the OS?

wouterverweirder, 2021-07-27:
Browser only: no. The only option to work with kinect data in the browser, is by using a nodejs (socket) server, which would act as a bridge between this library and the browser.

leskodan, 2021-07-27:
Okay, that makes sense. I assume the nodejs server would need to be running on the local client device? Essentially, some local application must be running to access Kinect and GPU?

wouterverweirder, 2021-07-28:
Not necessarily, you could have the kinect machine send the data to another machine using websockets. You could take a look at [Kinectron](https://kinectron.github.io/#/) an electron application which uses my library to send kinect data over a network.

leskodan, 2021-07-28:
Thanks @wouterverweirder , this is super helpful!

closed 2021-07-28 by leskodan