wouterverweirder/kinect-azure #53: Linux(Ubuntu) Build Support

challenger71498, 2021-07-30:
# Issues

I am using the repo with Linux kernel (Ubuntu 18.04) on my development environment, and just find out that it does not support build with Linux!

# Solution

So I just added Linux support on `binding.gyp`, so that `node-gyp` can build on both Windows and Ubuntu environement. PR is here #52.

---
And always, thank you for making this amazing asset. I really appreciate a lot :+1:

challenger71498, 2021-07-30:
# Another Issue

There was another issue though, so I drafted my PR.

When I tried to install the modified package to the other package on Linux environment, the following error occurs.

```
gyp: Undefined variable module_name in binding.gyp while trying to load binding.gyp
```

As far as I understand, in `package.json`, `install` script behaves as:
1. Tries to install pre-built `.node` file, but there is no such file that supports Linux environment, therefore next script is being executed.
2. If fails, tries to build with prebuild, but `prebuild` is dev-dependent, so with this case npm just skips it.
3. Rebuild with `node-gyp`.

Error message above tells that the third script is malfuctioning.
But I don't really know why this error occurs, since the package has been worked perfectly on Windows.
Even now, the original one (from your repo) does emit this same error message!

# Solution?

I guess if I move `prebuild` from `"devDependencies"` to `"dependencies"`, it will be work, because it somehow lets `node-gyp` to identify the `module_name` variable.

But I am kinda worried that it may destroy the mechanism of the install script, and I think package dependencies should be remain, if possible.

# Questions

I don't know the way how `node-gyp` gets module_name variable at `package.json`, to `binding.gyp`, so I cannot figure out why `node-gyp` cannot identify `module_name`.

So, mind if I have some questions:

1. How does `prebuild` and `node-gyp` related? By using `prebuild`, does it uses `node-gyp` internally?
2. If `prebulid` uses `node-gyp`, does `prebuild` passes `module_name` from `package.json` to `node-gyp` automatically? Then is there any reason should `prebuild` be dev-dependent?
3. If `prebuild` must be dev-dependent, is there any solution to reolve `module_name` variable to use it at `binding.gyp`?

---
Sorry for any impolite statement, since English is my second language.
Thank you :)

wouterverweirder, 2021-07-30:
Cool - would be nice to have Linux support! I think it might be better to move prebuild to dependencies, (instead of adding node-pre-gyp as an extra dependency).

challenger71498, 2021-07-30:
Great! I will search a bit if I can use `node-gyp` without moving `prebuild`, if cannot then I'll just move it, test it a bit more and open the PR again.

wouterverweirder, 2021-07-31:
Tested, worked like a charm :-) (I'm on Linux Mint 19 Cinnamon)
I've released as version 0.3.0 and created prebuilt binaries for both Linux and Windows.
Thanks a lot for this great contribution!

wouterverweirder, 2021-07-31:
if you want: @challenger71498 drop me an email at wouter.verweirder@gmail.com with your name, I'll add a thank you in the README file.

closed 2021-07-31 by wouterverweirder

challenger71498, 2021-07-31:
> Tested, worked like a charm :-) (I'm on Linux Mint 19 Cinnamon)
> I've released as version 0.3.0 and created prebuilt binaries for both Linux and Windows.
> Thanks a lot for this great contribution!

Glad to hear that it works fine! 
And also, I really appreciate that I could contribute to this awesome repo :D

> if you want: @challenger71498 drop me an email at [wouter.verweirder@gmail.com](mailto:wouter.verweirder@gmail.com) with your name, I'll add a thank you in the README file.

Sure! I'll send you an email soon.