wouterverweirder/kinect-azure #56: 0.3.0 problems with bodytracking

kristosb, 2021-09-01:
Hello, I tried to upgrade to 0.3.0 version from 0.0.19 and the body tracking doesnt work anymore. Other steams seem fine. I have the PATHs set up and dlls/onnx files copied. Can you verify on your side? Thanks

wouterverweirder, 2021-09-09:
Can you check if you are on version v1.1.0 of the Body Tracking SDK?

kristosb, 2021-09-09:
I checked versions:
v1.1.0 body tracking sdk 
v1.4.1 azure kinect sdk

If you are on the same setup maybe its some extra file needed copying somewhere. Would be good to see some list of the files including dlls and onnx in your project/module directory structure. I could make sure I have the same setup.

vilmire, 2021-11-10:
나도안돼 ㅜ

getarobo, 2021-11-10:
I am having a same problem, @kristosb  any luck fixing this?

kristosb, 2021-11-10:
I just use old version 0.0.19 of the wrapper.